galou/freecad.cross #4: Unable to open urdf or xacro with CROSS

ARLunan, 2023-11-25:
After installation in FreeCAD 21.1 on both MAC VM Ubuntu 22.04 aarch64 ROS 2 Humble & on native MAC M1 arm64 OS 17.1:
On MAC  VM  running g FreeCAD error message there is a "no module named "urdf_parser_py"
ON native MAC, while the Addon Manager reports installed and no error messages. 

But in both cases, unable to open .urdf or .xacro files .

galou, 2023-11-28:
Importing a URDF or xacro file requires ROS (i.e. calling `source /opt/ros/humble/setup.bash`,  for example, and launching FreeCAD from the same terminal).

What do you mean with he FreeCAD Custom Repository?

ARLunan, 2023-11-29:
I corrected the urdf_parser_py error by installing $ sudo apt install ros-humble-urdfom-py. I also copied the freeCAD launch file into ~/.appImage , i.e. "~/.appImage/FreeCAD_0.21.1-Linux-aarch64.AppImage . I can run freeCAD from anywhere with this path. SO where exactly in the ros sourced terminal that shows my ros2_ws workspace that contains my ros2 robot_description with urdf and xacro files do I run freCAD ?

ARLunan, 2023-11-29:
I tried your suggestion on my Ubuntu 22.04  ~/mtc_ws workspace which contains the "description" package urdf and xacro files.I sourced this mtc_ws workspace in the terminal from $ mtc_ws/ . install/setup.bash" .  I can display the description package name with the "$ ros2 pkg list|grep description" command. So I launch freeCAD that has the CROSS Addon installed in this terminal. If I Open & browse to the workspace src/...urdf directory or the workspace install/share ... urdf directory I cannot open a .xacro or .urdf file. Attahced is a capture of the freeCAD Open dialog noteing there is no .xacro or urdf is avaialnle to open in this list:
[freeCAD-cross-supported-formats.pdf](https://github.com/galou/freecad.cross/files/13505894/freeCAD-cross-supported-formats.pdf)

galou, 2023-11-30:
You cannot open any URDF or xacro file this way (I mean through FreeCAD's "Open..." command), you have to use the dedicated button in CROSS's toolbar. I know, the documentation is lacking (help welcome).

The reason for this (for now) is precisely the requirement that ROS must be sourced and the appropriate environment variables available in FreeCAD.

ARLunan, 2023-11-30:
After launching FreeCAD p[rogram (./FreeCAD_0.21.1-Linux-aarch64.AppImag ) from  a ROS sourced Terminal, navigating to Menu Bar>View>Workbench>CROSS able to open a  xacro file in a sourced linorobot2_ws robot workspace. Thank you as this should be very useful. I will create a "FeeCAD-CROSS Install & Run Cheatsheet". for my use and forward to you.

galou, 2023-12-01:
Great, thanks for this document in advance.

ARLunan, 2023-12-02:
Unfortunatly, the next day when I opened freecad.cross to load the same (and others in sourced termonals) previous URDF file, got this error:

buntutbc@macvm-ub22h-bv:~/.appImage$ Running the Python command 'UrdfImport' failed:
Traceback (most recent call last):
  File "/home/ubuntutbc/.local/share/FreeCAD/Mod/freecad.cross/freecad/cross/ui/command_robot_from_urdf.py", line 37, in Activated
    urdf_robot = UrdfLoader.load_from_file(filename)
  File "/home/ubuntutbc/.local/share/FreeCAD/Mod/freecad.cross/freecad/cross/urdf_loader.py", line 23, in load_from_file
    robot = Robot.from_xml_string(
  File "/opt/ros/humble/lib/python3.10/site-packages/urdf_parser_py/xml_reflection/core.py", line 612, in from_xml_string
    return cls.from_xml(node, path)
  File "/opt/ros/humble/lib/python3.10/site-packages/urdf_parser_py/xml_reflection/core.py", line 606, in from_xml
    return cur_type.from_xml(node, path)
  File "/opt/ros/humble/lib/python3.10/site-packages/urdf_parser_py/xml_reflection/core.py", line 228, in from_xml
    obj.read_xml(node, path)
  File "/opt/ros/humble/lib/python3.10/site-packages/urdf_parser_py/xml_reflection/core.py", line 594, in read_xml
    self.XML_REFL.set_from_xml(self, node, path)
  File "/opt/ros/humble/lib/python3.10/site-packages/urdf_parser_py/xml_reflection/core.py", line 528, in set_from_xml
    raise ParseError(e, path) # get_attr_path(attribute.xml_var)

ParseError in /robot:
ubuntutbc@macvm-ub22h-bv:~/.appImage$

galou, 2023-12-06:
Eventually, the [issue](https://github.com/turtlebot/turtlebot4/pull/320) was in the `turtlebot4_description` package.

jetdillo, 2024-01-16:
I am having a similar problem. 
Running FreeCAD 0.21.2 on Ubuntu 20.04, ROS Noetic. 
I've cloned the repo here, installed:
```
ros-noetic-urdfdom-py
urdf-parser-py (via pip) 

``` 
 and followed the install instructions. 
 I still get 
`No module named 'urdf_parser_py'`

When starting FreeCAD and then when I try to load in a .xacro file:
```
Running the Python command 'UrdfImport' failed:
Traceback (most recent call last):
  File "/home/armadilo/.local/share/FreeCAD/Mod/freecad.cross/freecad/cross/ui/command_robot_from_urdf.py", line 37, in Activated
    urdf_robot = UrdfLoader.load_from_file(filename)

name 'UrdfLoader' is not defined
```

I have ROS Noetic installed and source my workspace from my .bashrc. 

I can import urdf_parser_py from the REPL:
`(base) armadilo@roadrunner:~/projects/freecad.cross$ python
Python 3.10.9 (main, Mar  1 2023, 18:23:06) [GCC 11.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import urdf_parser_py
>>> dir(urdf_parser_py)
['__builtins__', '__cached__', '__doc__', '__file__', '__loader__', '__name__', '__package__', '__path__', '__spec__']
>>>`

I'm pretty new to FreeCAD, but quite familiar with ROS. 
I think there's a lot of promise to this and would really like to use it to help with a Gazebo project I'm currently working on.

galou, 2024-01-16:
ROS1 is not supported.

SimeonOA, 2024-06-17:
Hi,

I am having this same issue when trying to load a urdf file.

Here are my setup details:
Computer: Ubuntu 22.04
ROS Version: Humble

When I run ./FreeCAD.AppImage --appimage-extract-and-run, the image below is my result:

[image]

When I try to import the urdf file, I get this error:

[image]

I have tried ` pip install urdfpy`, `pip install urdf-parser-py` and `sudo pip install urdf-parser-py`.
I have also tried adding `~/.local/lib/python3.10/site-packages` to the `PYTHONPATH`. 
None of these work.
Do you have any suggestion on what I could do?
Thanks

galou, 2024-06-17:
Did you try to source a ROS workspace?

SimeonOA, 2024-06-21:
Yes. I sourced my ROS workspace. Also after running colcon build, I sourced the install/setup.bash.

I am still having the error.

galou, 2024-06-24:
Can you run `python -c "import urdf_parser_py"` in the same terminal where you launch FreeCAD?

drfenixion, 2024-07-10:
> Yes. I sourced my ROS workspace. Also after running colcon build, I sourced the install/setup.bash.
> 
> I am still having the error.

Also you can try follow https://github.com/galou/freecad.cross/tree/main/docker README.md
It will run FreeCAD + CROSS inside docker container and also install needed dependencies via helper ROS package.
Or just install dependencies from that package to your system directly via "rosdep install" (see above readme).

galou, 2024-08-01:
I'll close this issue in case @SimeonOA doesn't react.