galou/freecad.cross #14: Need demonstration or tutorial how to use "Set placements" tool

drfenixion, 2024-02-18:
I trying to use "Set placement" tool various ways but without any usefull result because I dont undastand how it works.
I cant use freecad.cross without this tool because not way to change position of joint origin.
Need any demo or tutorial for this "Set placements" tool.
Also it is good to have visual tool like regular Transform (how we do with bodies) for change origin of Joint.

galou, 2024-02-19:
I totally agree with you that the tool is difficult to use and that a video tutorial is something that is missing. I'll try to find some time to create one.

In the meantime, you can use the three dots on the right of the "Origin" property to set the origin with a visual feedback (by clicking on "Apply").

I know a mean to use the regular Transform tool but it requires that the parameter is called "Placement". Now, the property is called "Origin" to fit the URDF jargon and the placement is the absolute joint position, a read-only property.

drfenixion, 2024-02-21:
> In the meantime, you can use the three dots on the right of the "Origin" property to set the origin with a visual feedback (by clicking on "Apply").

Thanks, forgot about this option but with that way it will long job for big models to set joints origins. it's also not convenient.

I tried to do fast fix for add transform tool but i dont have experience with FreeCAD dev and need your opinion. 

I tried to use App::GeometryPython instead of App::FeaturePython in joint_proxy it looks same but with geometric fetures. Added placement property. 
That appear transform tool in menu, but tool does not work (without any errors, just does not appear in 3d space).
Also tried various extensions for App::GeometryPython. They also does not give working transform tool.

I tried other objects like App:Part that give working transform tool but it is not right way because does have not proxy property and not compatible with code.

Any key that i should to use in code for get working transform tool and also have proxy for compatible?

My idea is adding transform tool with something like App::GeometryPython (instead of App::FeaturePython in joint_proxy.py) and when it will change placement do sync with origin (for compatible). Also it may be need to fix ciclic update placement<->origin (if will).

drfenixion, 2024-02-25:
Added related issue [https://github.com/galou/freecad.cross/issues/21](https://github.com/galou/freecad.cross/issues/21)

closed 2024-02-25 by drfenixion

drfenixion, 2024-03-05:
> Cf. https://github.com/galou/freecad.cross/wiki/Command-SetPlacement
> 
> Please feel free to try it out and improve the documentation.

Thank you. 

With "Setting the MountedPlacement property of links" docs is ok.

But instructions "Setting the Origin of joints" does not clear for me. I was experimenting again and find only working way how to set origin to right position with "SetPlacement" command.
Working way is "select joint, select child link LCS (where it should be connected to parent link), select parent link LCS (where it should be connected to child link). After that child link will setted to the parent link LCS (for child link connection).

My worked result (child LCS established in parent LCS for child link as I wanted)  
[image]

It will nice if you rewrite 2 lines from wiki - 
"a CROSS::Joint, the LCS of the parent link, and the LCS of the child link on the same link,
a CROSS::Joint, the "child" LCS on the parent link, and the "parent" LCS on the child link."
with other words and that will clear for me (maybe not only me). It will give me opportunity to use other ways for setting joint position.