galou/freecad.cross #15: Tips for usinkg CROSS for custom build ros with --symlink-install

sprhawk, 2024-02-24:
I built `ROS2` under `debian bookworm` with `colcon build --symlink-install`

All the ros packages are installed in separate directory hierarchy , and `ros_workspace/install/setup.bash` will set `PYTHONPATH` for every packages. But because `FreeCAD` will erase environment variable  `PYTHONPATH` after startup,  I provide here a solution for that:

```
FreeCAD  -P ${PYTHONPATH//:/' -P '}
```

BTW: the required `libnglib` on `Debian Bookworm` is installed under /usr/lib/x86_64-linux-gnu/netgen, which is not found by FreeCAD. So I need to add
```
export LD_LIBRARY_PATH=/usr/lib/x86_64-linux-gnu/netgen:$LD_LIBRARY_PATH
```
to start FreeCAD + freecad.cross properly

drfenixion, 2024-02-25:
Maybe you should add PR with TROUBLESHOOTING.md with short description of this case?
@galou, what do you think?

galou, 2024-02-29:
I added a remark in the `README.md` file, cf f400ae4. Thanks!

galou, 2024-02-29:
What is the nglib library required for?  Is it really required by CROSS because it's not installed on my system but CROSS obviously works?

galou, 2024-03-01:
Maybe related to [this](https://forum.freecad.org/viewtopic.php?p=731355#p731355)?

sprhawk, 2024-03-03:
> Maybe related to [this](https://forum.freecad.org/viewtopic.php?p=731355#p731355)?

My Debian just install `nglib` under a `netgen` subdir which cannot be found by FreeCAD

galou, 2024-03-04:
If this is required by ROS, then `setup.bash` should set `$LD_LIBRARY_PATH` correctly. Isn't it the case?

sprhawk, 2024-03-08:
> If this is required by ROS, then `setup.bash` should set `$LD_LIBRARY_PATH` correctly. Isn't it the case?

My bad. the netgen is required by FreeCAD,  not ROS

closed 2024-03-08 by galou