galou/freecad.cross #18: missing FeaturePython caused python exception

sprhawk, 2024-02-25:
https://github.com/galou/freecad.cross/blob/a138f1aa482b3da99e6ccb096891cf947f72e65a/freecad/cross/robot_proxy.py#L668

```
PropertyPythonObject::toString(): failed for <class 'freecad.cross.robot_proxy.RobotProxy'>
Traceback (most recent call last):
  File "/usr/lib/python3.11/json/__init__.py", line 231, in dumps
    return _default_encoder.encode(obj)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/json/encoder.py", line 200, in encode
    chunks = self.iterencode(o, _one_shot=True)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/json/encoder.py", line 258, in iterencode
    return _iterencode(o, 0)
           ^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/json/encoder.py", line 180, in default
    raise TypeError(f'Object of type {o.__class__.__name__} '
<class 'TypeError'>: Object of type FeaturePython is not JSON serializable
PropertyPythonObject::toString(): failed for <class 'freecad.cross.link_proxy.LinkProxy'>
Traceback (most recent call last):
  File "/usr/lib/python3.11/json/__init__.py", line 231, in dumps
    return _default_encoder.encode(obj)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/json/encoder.py", line 200, in encode
    chunks = self.iterencode(o, _one_shot=True)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/json/encoder.py", line 258, in iterencode
    return _iterencode(o, 0)
           ^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/json/encoder.py", line 180, in default
    raise TypeError(f'Object of type {o.__class__.__name__} '
<class 'TypeError'>: Object of type FeaturePython is not JSON serializable
PropertyPythonObject::toString(): failed for <class 'freecad.cross.link_proxy.LinkProxy'>
Traceback (most recent call last):
  File "/usr/lib/python3.11/json/__init__.py", line 231, in dumps
    return _default_encoder.encode(obj)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/json/encoder.py", line 200, in encode
    chunks = self.iterencode(o, _one_shot=True)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/json/encoder.py", line 258, in iterencode
    return _iterencode(o, 0)
           ^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/json/encoder.py", line 180, in default
    raise TypeError(f'Object of type {o.__class__.__name__} '
<class 'TypeError'>: Object of type FeaturePython is not JSON serializable
PropertyPythonObject::toString(): failed for <class 'freecad.cross.link_proxy.LinkProxy'>
Traceback (most recent call last):
  File "/usr/lib/python3.11/json/__init__.py", line 231, in dumps
    return _default_encoder.encode(obj)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/json/encoder.py", line 200, in encode
    chunks = self.iterencode(o, _one_shot=True)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/json/encoder.py", line 258, in iterencode
    return _iterencode(o, 0)
           ^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/json/encoder.py", line 180, in default
    raise TypeError(f'Object of type {o.__class__.__name__} '
<class 'TypeError'>: Object of type FeaturePython is not JSON serializable
PropertyPythonObject::toString(): failed for <class 'freecad.cross.link_proxy.LinkProxy'>
Traceback (most recent call last):
  File "/usr/lib/python3.11/json/__init__.py", line 231, in dumps
    return _default_encoder.encode(obj)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/json/encoder.py", line 200, in encode
    chunks = self.iterencode(o, _one_shot=True)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/json/encoder.py", line 258, in iterencode
    return _iterencode(o, 0)
           ^^^^^^^^^^^^^^^^^
[... 301 lines cut ...]
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/json/encoder.py", line 200, in encode
    chunks = self.iterencode(o, _one_shot=True)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/json/encoder.py", line 258, in iterencode
    return _iterencode(o, 0)
           ^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/json/encoder.py", line 180, in default
    raise TypeError(f'Object of type {o.__class__.__name__} '
<class 'TypeError'>: Object of type ViewProviderDocumentObject is not JSON serializable
PropertyPythonObject::toString(): failed for <class 'freecad.cross.link_proxy._ViewProviderLink'>
Traceback (most recent call last):
  File "/usr/lib/python3.11/json/__init__.py", line 231, in dumps
    return _default_encoder.encode(obj)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/json/encoder.py", line 200, in encode
    chunks = self.iterencode(o, _one_shot=True)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/json/encoder.py", line 258, in iterencode
    return _iterencode(o, 0)
           ^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/json/encoder.py", line 180, in default
    raise TypeError(f'Object of type {o.__class__.__name__} '
<class 'TypeError'>: Object of type ViewProviderDocumentObject is not JSON serializable
```

drfenixion, 2024-02-25:
I did see that "toString(): failed" when saving doc. I dont know that it broke but looks saving works. 
Can you fix that error?

sprhawk, 2024-02-26:
> I did see that "toString(): failed" when saving doc. I dont know that it broke but looks saving works. Can you fix that error?

I'm still not familiar with FreeCAD plugin dev, I need to learn why use FeaturePython, then can give a solution for that.

sprhawk, 2024-02-27:
> I did see that "toString(): failed" when saving doc. I dont know that it broke but looks saving works. Can you fix that error?

I have no clue how to debug it. It seems set a property to JointProxy or ViewProviderDocumentObject with `TypeError`, but I don't know which is.

galou, 2024-02-27:
This JSON serialization is something new and I still need to figure out how to solve it. Help welcome!

drfenixion, 2024-04-04:
@galou 
Looks like it is related to https://forum.freecad.org/viewtopic.php?t=84305
and can be solve by change FC version to 0.21.1 or latest_dev_build or 0.21.3 (in future).

galou, 2024-04-16:
User of FreeCAD <= 0.21.1 should use https://github.com/galou/freecad.cross/releases/tag/fc_v0.21.1.

A few more commits may be compatible but I'm going to remove the `__setstate__` and `__getstate__` functions.

closed 2024-04-16 by galou