galou/freecad.cross #19: hard coded Mod path prevent from starting via FreeCAD -M argument

sprhawk, 2024-02-25:
https://github.com/galou/freecad.cross/blob/a138f1aa482b3da99e6ccb096891cf947f72e65a/freecad/cross/wb_utils.py#L34

due to this hard coded UserAppData Path, I cannot start `FreeCAD` with `cross` via `FreeCAD -M freecad.cross` command line

sprhawk, 2024-02-29:
> It's not hard-coded since it uses `getUserAppDataDir()`. What is the error message? Why do you need to pass an additional module path. You can install CROSS like any other FreeCAD 

I used debian with no system provided ROS2, and custom built ROS workspace ( so it is not in system path ), and CROSS requires ROS2 environment setup during start up, I don't want it loaded when I don't uses cross ( don't setup ROS2 environment)

galou, 2024-02-29:
CROSS is supposed to load also without ROS. If it doesn't, it's a bug.

Would you otherwise have a solution for this issue ?