galou/freecad.cross #26: There is bug calculating robot link positions in robot

sprhawk, 2024-02-27:
I'm using this [model](https://github.com/unitreerobotics/unitree_ros/blob/master/robots/go2_description/urdf/go2_description.urdf) or [this one](https://github.com/unitreerobotics/unitree_ros/blob/master/robots/go2_description/xacro/robot.xacro)

[image]

Left one is result of `import urdf or xacro`, right one is `create assembly from urdf or xacro`

And I try to use `Set Placement` to move some additional models  via assembly LCS, but it seems the calculation of Placement is based on parent's Origin, not considered on hierarchy of Links. ( links of transformation of Placements )

galou, 2024-02-28:
Do you have a ROS2 port of the package so that I don't need to port it myself?

sprhawk, 2024-02-28:
> Do you have a ROS2 port of the package so that I don't need to port it myself?

[Uploading edi_arm_description.zip…]()

Take a try.

sprhawk, 2024-02-28:
> The link to the file is broken.

updated again

galou, 2024-02-28:
I can open `robot.xacro` by importing it as a Robot but the files `go2_full.urdf` and `go2.urdf` cannot be opened in CROSS because they cannot be parsed by the underlying library `urdf_parser_py`. The issue is the presence of `<?xml version="1.0" encoding="utf-8"?>`. Remove it to make `urdf_parser_py` happy.

I didn't find the file `go2_description.urdf` in your archive.

sprhawk, 2024-02-29:
> I can open `robot.xacro` by importing it as a Robot but the files `go2_full.urdf` and `go2.urdf` cannot be opened in CROSS because they cannot be parsed by the underlying library `urdf_parser_py`. [...]

Dose it show in correct position ? like the image I posted in right ?

galou, 2024-02-29:
Yes.