galou/freecad.cross #27: Compatibility Issues with CROSS Workbench in Freecad

AthidhiVijai, 2024-02-29:
Dear Developers,

I am writing to express my appreciation for the development and release of the CROSS workbench for Freecad. It seems like a valuable and powerful tool.

As a learner of ROS2 Humble and Freecad, I have followed the installation instructions and successfully installed CROSS in various Freecad versions (0.21.2, various versions within 0.22, and Freecad realthunder). While I am able to utilize CROSS functionality in Freecad 0.22 (both versions), it is not available in Freecad 0.21.2.

However, I have encountered an error related to NumPy in the Curves workbench when working on robot design in Freecad 0.22. This error does not occur in Freecad 0.21.2, but the CROSS workbench is not accessible in this version.

I am seeking your assistance in navigating this situation. I would be grateful if you could provide guidance on the following:

    Potential solutions to address the compatibility issue with CROSS in Freecad 0.21.2. Are there any settings or configurations that can enable its functionality or workarounds for using CROSS features in this version?
    Troubleshooting advice for the NumPy error encountered in the Curves workbench within Freecad 0.22.

Thank you for your time and consideration. I look forward to your insights and assistance.

Sincerely,
Vijai

galou, 2024-02-29:
Hi Vijai, I added a notice in the `README.md` file about launch FreeCAD with ROS. Please have a look. I successfully tested both 0.22 and 0.21.

Can you please launch FreeCAD from the command line and report any error related to CROSS. Please precise which OS and what variant of FreeCAD you use (AppImage, package manager, built from source, ...).

AthidhiVijai, 2024-02-29:
Thank you for your immediate response,
okay i will go through README.md
and i am Using Ubuntu 22.04
and Freecad Downloaded from snap store/Softwarecenter/Ubuntu software and
version i tried all version available in Ubuntu Software center , it worked
only in 0.22 is the best not in 0.21.2
thank you

galou, 2024-03-01:
Sorry, I didn't get your answer.

AthidhiVijai, 2024-03-12:
Thank you for your prompt reply. I'm very interested in creating a tutorial video showcasing the capabilities of this workbench.

To ensure the video is comprehensive and beneficial for viewers, I'd appreciate some guidance on the functionalities of each tool included in the workbench.  Following the 4W-1H framework (What, When, Where, Why, How) would be particularly helpful.

With this information, I'm confident I can create a valuable YouTube video that will empower users and make them comfortable using the workbench to its full potential.

Thank you for your time and consideration.