galou/freecad.cross #29: Box (sphere) from bounding box - "No compatible object selected" (when used link as source)

drfenixion, 2024-03-10:
I tried to build bounding box from box and get error if i use Link as source. Link from another document.
Any not little project uses a lot of linked documents as inserted object in main docs. It will be nice to fix that bug.

galou, 2024-03-15:
I'm not sure that it's not a good idea to use link because links usually have a different placement than the linked part. The generated object will then be at a different place than the link and this may confuse users. In FreeCAD, you can jump to the linked part with a right-click on the link in the tree.

drfenixion, 2024-03-16:
Maybe there can be wayaround. Because as i know linked documents only way to work with big projects and also team project dev (different developers makes different parts and uses parts from each other as linked documents).
I am ok with current workflow but looks it is lack when project scale.