galou/freecad.cross #32: Wrong description of "Set placement" tool

drfenixion, 2024-03-12:
"a CROSS::Joint, the "child" LCS on the parent link, and the "parent" LCS on the child link"
If do that joint will be placed in wrong position.
Only working way is a "CROSS::Joint, the "parent" LCS on the child link, and the "child" LCS on the parent link".
Thus should be changed the selection of links LCS.

It is the reason why i couldn't get any useful result before just experimenting.

"CROSS::Joint, the LCS of the parent link, and the LCS of the child link on the same link," - this works ONLY for "centered" supporting LCS (their center positions on 0,0 coordinate of basic axies). 
For example if you use cude (from Part workbench) which have 0.0 coordinate on one of  his angles (and thos body not "centered") and set LCS on his edge center this selection tip will works wrong (join will be placed in wrong position).
This should be changed to -
"CROSS::Joint, the LCS of the parent link, and the LCS of the child link on the same link (works only for centered supporting LCS, you must place supporting LCS on 0.0 coordinate of basic axis)," 

Result correct tips:
"CROSS::Joint, the "parent" LCS on the child link, and the "child" LCS on the parent link".
"CROSS::Joint, the LCS of the parent link, and the LCS of the child link on the same link (works only for centered supporting LCS, you must place supporting LCS on 0.0 coordinate of basic axis),"

drfenixion, 2024-03-13:
Because of so many rules and preparation of using current Set placement tool. It nice to make new tool that just will connect LCS of child and LCS of parent link without caring of centering LCS of child to origin point. Looks that can be achieved using other math of current Set placement.
@galou , what do you think?

galou, 2024-03-14:
There is actually no need to move objects around. Please let me some little time and I'll improve the wiki with the SetPlacement tutorial for joints.

galou, 2024-03-15:
Can you please try the [mini tutorial about SetPlacement for joints](https://github.com/galou/freecad.cross/wiki/Command-SetPlacement)?

drfenixion, 2024-03-15:
> Can you please try the [mini tutorial about SetPlacement for joints](https://github.com/galou/freecad.cross/wiki/Command-SetPlacement)?

Thank you.
I tryed. It is working . But docs have some actions missed between instructions (i know what todo but newbe can stack).
Also it is not a good way to teach "Setting the MountedPlacement" and "Setting the Origin of joints" on differents projects.

I believe it should be one simple project (without redundant parts, it looks the robot_5r is will be nice to adapt) and steps should be:
 1) "Setting the MountedPlacement"
 2) "Setting the Origin of joints"
 for complete picture of needed actions todo.
 
 About my issue. Looks i forgot  "Setting the MountedPlacement" step. Maybe that is reason of necessity centuring LCS to origin of bodies. I just did only "Setting the Origin of joints". I will try to reasamble my project with "Setting the MountedPlacement" step also.

drfenixion, 2024-03-18:
**Tips for newbe**. 

It is better to use centered bodies (Mounted placement and that (parent - where is link should be connected to his parent link) LCS at 0.0 coordinate of axis). 

Also is better to use same type of LCS attachments (Inertial attach to face is well suited). 
If you you will use diffent LCS attachment types it can confuse you then you will use "Set Placement" tool becuse you will need to search currect combination of selection (child-parent link, child-parent LCS, etc).

This tips will let you do less actions of setting Set placement of Joints, links. 

Worked ways to use "Set Placement" tool:
### 1)
**Set mounted placement of child link.**
Select: child link, parent link, parent LCS of child link (where is child should be connected to parent link)). Click to Set Placement tool. Your Mounted placement will setted to your LCS. It is needed to positioning joint inside child link.
_If you use centered links with "mounted placement" LCS in 0.0. and needed angle of body at modeling stage this step no need._

**Set joint placement.**
Select: join, parent LCS of parent link, child LCS of parent link (where parent should be connected to child link). Click to Set placement tool.

### 2)
**Set mounted placement of child link.**
Same as in firts way.

**Set joint placement.**
Select: join, parent LCS of child link, child LCS of parent link (where parent should be connected to child link). Click to Set placement tool.

**Set mounted placement of child link.**
Select: parent link, child link, parent LCS of child link (where is child should be connected to parent link)). Click to Set Placement tool. 
_This step may be not necessary if child link is centered and corrent oriented._

How you can see first way is shorter.

closed 2024-04-08 by drfenixion