galou/freecad.cross #37: Auto calculating of mass and inertia

drfenixion, 2024-03-19:
I am sure we need auto calc of mass and inertia for links. I did mass calc sript for mass for Blender Phobos. Phobos (how i remember) can calc inertia but you need to set mass manually for it. 
I used my script for it for auto add mass to all links and after used Phobos utils for calc inertia based on my mass.

My idea adapt my mass calc script for Freecad and pull Phobos inertia calc script and also adapt it together.
Add option to choose density of link (or a whole robot). Density can be finded in inet for any material.
Add button of calc mass and inertial of all links based on density.
@galou , what do you think? Any suggestion what can help.

**My mass calc script for Blender.**
```
import bpy
from bpy import context

# Quick way to test with/without hidden faces
USE_FILTER_FACES = True
object_density = 1040 # ABS plastic
objs = bpy.context.view_layer.objects.selected

def is_face_skip(f):
    """Ignore faces that pass this test!"""
    return f.hide is False
    # You may want to filter based on material.
    # return f.material_index == 0
    
def bmesh_from_object_final(ob):
    import bmesh
    matrix = ob.matrix_world
    me = ob.to_mesh()
    me.transform(matrix)
    bm = bmesh.new()
    bm.from_mesh(me)
    if USE_FILTER_FACES:
        faces_remove = [f for f in bm.faces if not is_face_skip(f)]
        for f in faces_remove:
            bm.faces.remove(f)
    return (bm, matrix.is_negative)

def volume_and_area_from_object(ob):
    bm, is_negative = bmesh_from_object_final(ob)
    volume = bm.calc_volume(signed=True)
    area = sum(f.calc_area() for f in bm.faces)
    bm.free()
    if is_negative:
        volume = -volume
    return volume, area

def get_mass(density, volume):
    return density * volume

for ob in objs:
    volume_single, area_single = volume_and_area_from_object(ob)
    ob["mass"] = get_mass(object_density, volume_single)

```

drfenixion, 2024-03-20:
@galou, I found macro that can calc all mass and ineria. It need adapt code and make tool in CROSS for auto calc that for links based on density (optionaly maybe add material option choose from).

FCInfo macro.
[image]
[image]

drfenixion, 2024-03-20:
#workpiece

#calc of mass
density = 1040 # ABS plastic
obj_name = 'Body002'
model = FreeCAD.ActiveDocument.getObject(obj_name)
volume = model.Shape.Volume
mass = volume * density

#calc of inertia
# todo

drfenixion, 2024-03-20:
Started implementation.

galou, 2024-03-21:
This is missing from the current tool, for sure. I'm not convinced that you can get anything better than an approximation and it's a lot of work to implement this because objects in FreeCAD tend to be inconsistent (mesh vs. shape, placement of links, objects without shape, ...).
I'm more concentrated on implementing functions to debug a running system so I won't have time for the modeling part of the workbench. I'll glad if you come with an implementation.

drfenixion, 2024-04-02:
@galou 
Looks like I'm close to the result with mass and inertia auto calculation.
Todo: 
1) To bug fix - some elements have orientation drifting. looks likes it depend of as body was modeled. Most oriented well.
2) Minor fixes and improvement.

You can see inertia blocks. They must be aproximary same size and position/orientation as their links.
2 inertia blocks in model has orientation drift, others are ok. Size and position are ok of all blocks.
[image]

drfenixion, 2024-04-08:
Created fork named OVERCROSS with:

1. Material selection of link or whole robot
2. Auto calculation of mass and inertia
3. Launcher for Gazebo
4. New some icons and icons reordering
5. Bug fixes
6. Others

https://github.com/drfenixion/freecad.cross

closed 2024-04-08 by drfenixion