galou/freecad.cross #41: Lack of tutorial: I can't confidently use the Cross workbench

MJavadZallaghi, 2024-07-12:
I was trying to generate URDF using original Cross workbench for FreeCAD, but as there was no tutorial, I can not continue using the tool. Indeed, I'm confused how to configure joints and positioning them using LCF (local coordinate frames) of the parts I've crated.

So I've created this issue to ask:

- Is there any tutorial to let me know how I can confidently use it? I'll appreciate your valuable support!

MJavadZallaghi, 2024-07-13:
Thanks to [this tutorial](https://vk.com/@robotforge-freecad-cross-open-source-dlya-robotov), I was able to convert my robot designed in FreeCAD using CROSS to a ROS 2 package and Gazebo!
[image]
[image]
To be honest, the CROSS benchwork you've developed is awesome. Thanks for creating it!

closed 2024-07-13 by MJavadZallaghi

galou, 2024-08-01:
I know about this issue. The good point is that you know have the capacity to write a tutorial in English ;)

The only tutorial I have now is the [Github Wiki](https://github.com/galou/freecad.cross/wiki).