galou/freecad.cross #45: Material editor bug

johnny555, 2024-09-17:
Hey, 

I'm using FreeCAD 1.0.0 RC1 and when I try to add mass and inertia to my robot I get the following error: 

```
10:03:51  Running the Python command 'CalculateMassAndInertia' failed:
Traceback (most recent call last):
  File "C:\Users\johnv\AppData\Roaming\FreeCAD\Mod\freecad.cross\freecad\cross\ui\command_calculate_mass_and_inertia.py", line 48, in Activated
    default_material = material_from_material_editor(robot.MaterialCardPath)
                       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\johnv\AppData\Roaming\FreeCAD\Mod\freecad.cross\freecad\cross\freecad_utils.py", line 738, in material_from_material_editor
    material_editor = MaterialEditor.MaterialEditor(card_path=card_path)
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "C:\Users\johnv\Downloads\FreeCAD_1.0.0RC1-conda-Windows-x86_64-py311\FreeCAD_1.0.0RC1-conda-Windows-x86_64-py311\Mod\Material\MaterialEditor.py", line 72, in __init__
    self.widget = FreeCADGui.PySideUic.loadUi(filePath + "Resources" + os.sep + "ui" + os.sep + "materials-editor.ui")
                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<string>", line 4, in <module>
```

I've found a ticket on the main FreeCAD branch that is related, but just also putting this here in case this is a CROSS specific bug. 

https://github.com/FreeCAD/FreeCAD/issues/16516

drfenixion, 2024-09-21:
It is not related to CROSS.

There is a fix https://github.com/FreeCAD/FreeCAD/pull/16322

I think i will be corrected after fix merged.

drfenixion, 2024-09-24:
Now it is fixed in FreeCAD 1.0rc2. Can be closed.

johnny555, 2024-09-27:
Thanks! Yep its fixed now in rc2 !

closed 2024-09-27 by johnny555