galou/freecad.cross #49: Is Cross ready for FreeCAD 1.0 Release ?

luzpaz, 2024-10-31:
* https://github.com/FreeCAD/FreeCAD-addons/issues/305

galou, 2024-10-31:
I guess yes since I'm using the `freecad-daily` package on Ubuntu:
```
OS: Pop!_OS 22.04 LTS (sway-nvidia/xcb)
Architecture: x86_64
Version: 1.1.0dev.39096 (Git)
Build type: Release
Branch: main
Hash: 4f323f9580334de25192d2bf1f272206ad2d3f55
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 9.1.0, OCC 7.8.1
Locale: C/Default (C) [ OS: English/United States (en_US) ]
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Qt default
Installed mods: 
  * freecad.cross 0.5.0
  * Assembly4 0.50.15
  * boltsfc 2022.11.5
  * freecad.gears 1.3.0
  * fasteners 0.5.29
  * Freecad-Built-in-themes-beta 1.2.2
  * OpenTheme 2024.9.1
  * Defeaturing 1.2.2
  * RobotDescriptor
  * FreeCAD-AI-Toolbar
  * MeshRemodel 1.10.32
  * MnesarcoUtils 0.2.14
```

galou, 2024-10-31:
I could not test with Qt6 because I cannot compile:
```
/home/gael/04-sources/freecad/FreeCAD/src/Gui/PythonWrapper.cpp:131:11: fatal error: signalmanager.h: No such file or directory
  131 | # include <signalmanager.h>
```

I also tested Coin3D 4.0.3 but I cannot launch FreeCAD with it. Cf. https://github.com/coin3d/coin/pull/531#issuecomment-2392944875.

luzpaz, 2024-10-31:
```
/home/gael/04-sources/freecad/FreeCAD/src/Gui/PythonWrapper.cpp:131:11: fatal error: signalmanager.h: No such file or directory
  131 | # include <signalmanager.h>
```

Any chance you'd be willing to open an upstream FreeCAD ticket for this ?

closed 2024-10-31 by luzpaz