galou/freecad.cross #60: Problems with link name list

mosfet80, 2025-03-10:
[image]

The names of the links in the tree do not match those in the drop-down menu.

I use freecad 1.1 dev

DiuLaMaX, 2025-03-12:
I also have similar issue. I created a link called base link, but it only showed as `Link` when I created a new joint. Even if I created multiple links and joints with different label, they were all named `link` and `joint` when I output them into URDF. Do I missed any step when I create the joint and link? Also, I got this warning when I opened FreeCAD. `No module named 'urdf_parser_py'
`
[image]

galou, 2025-03-13:
I'll have a look.

In the meantime I can answer that `No module named 'urdf_parser_py'` is unrelated. The reason for this warning is that no ROS workspace was sourced before launching FreeCAD (you have to launch FreeCAD from the command line if you want to use ROS functionalities (or of course that `ros-humble-urdfdom-py` or similar is not installed).

galou, 2025-03-13:
@DiuLaMaX You have only one link in your robot. You should also set the description to `base_link`, this has priority over the label. The reason for this is that FreeCAD doesn't allow duplicated labels by default but two different robots should be allowed to have the same link or joint names. As I wrote, I'll change that to a normal propriety.